bverc/retropie-status-overlay #16: Added MCP chip support, wont PR because wont merge due to other changes

louisvarley, 2020-10-12:
Feel free to pull anything from my build. 

Added MCP chip support
Voltage multiplier so you can narrow down your ADC to voltage 
Disable or enable in-game overlay
Disable shutdown on low battery (mainly for testing or to stop loops)

https://github.com/louisvarley/retropie_status_overlay

Thanks pal

louisvarley, 2020-10-12:
Oh and i also...
- changed the install.sh to create a service and auto install. 
- It could still be installed by just downloading install.sh with some tweaks 
- Can be installed from anywhere, picks up script_path and installs service using that path

still not sure service name "overlay" is unique enough to not clash with something else.

bverc, 2020-10-12:
Wow nice work. Had a quick look and looks like some good improvements. Will test out when I next get a chance and will likely pull many through.

Cheers.

bverc, 2020-10-12:
Open to name ideas too? Overlay was the term used by @drez but I agree it clashes with Theme Overlays. Also keen to replace underscores with hyphens.

louisvarley, 2020-10-12:
> 
> 
> Open to name ideas too? Overlay was the term used by @drez but I agree it clashes with Theme Overlays. Also keen to replace underscores with hyphens.

retropie-status-overlay would probably work. If it works for you i'm happy to get it into a PR'able state. Still testing that the MCP chip works correctly. It's not as reliable as the ADS1 but i could'nt fit one into my Game Gear shell.

louisvarley, 2020-10-12:
Also couldn't test to ensure ADS1 still works as i don't have the chip.

louisvarley, 2020-10-13:
changed service name, had to change VMIN and VMAX to get working on mine as expected, so i moved to config.ini

bverc, 2021-02-07:
Finally having a chance to look at this. I've reviewed your code and identified the following issues:

- The Shutdown function no longer respects the BatteryLDO setting as the check for ADCShutdown was inside the shutdown function. It should instead move to the ADC checking before shutdown() is called.
- Removed the manual copying of config.ini.example if during install, Default config is chosen
- Installation of service is failing with following error:

> Failed to start retropie-status-overlay.service: Unit retropie-status-overlay.service has a bad unit file setting.

Most likely this is due to the double slash in the following:

> sed: can't read /lib/systemd/system**//**home/pi/retropie_status_overlay/retropie-status-overlay.service: No such file or directory

I'll have a go at cleaning up these issues in a separate branch, then once I'm happy I'll merge into the master branch.

bverc, 2021-02-07:
Ok so I have created branch:
https://github.com/bverc/retropie_status_overlay/tree/louisvarley-edits

Further to the fixes to the issues identified above, the first commit also includes:

- Removal of trailing spaces
- Removal of tabs which screwed up formatting
- Restoration of previous default values
- Correction of overlay service name in install.sh
- Added manual install instructions back into README
- Removed all MCP ADC specific content from README. Not sure where this should live long term, potentially in a wiki?

I'll test this out on a few more devices and once I'm happy will merge into master branch

bverc, 2021-02-07:
Why does the entire install script need to be run as root?

Surely it would be better to just rely on sudo to create the service at the end and to copy pngview to /usr/local/bin/. Otherwise root owns everything, and you cant even edit config.ini without sudo.

bverc, 2021-03-04:
Merged into master. Will leave this issue open until @louisvarley can comment.

louisvarley, 2021-03-04:
Thanks @bverc sorry i've not been able to help with the merge, lot of covid related personal issues last few months. Is there anything additional you need from me?

bverc, 2021-03-04:
No problem. Nothing specific required. All seems to be working ok. If you get a chance, and have look at what I took and what I left, and see if it makes sense. All your commits are in the history, but might break your fork if you pull again. 

Also see question above regarding sudo.

bverc, 2022-05-03:
@louisvarley I have modified to modularise the adc functions. This abstracts out to an import based on the text used in the config file, allowing any number of possible adc modules to be added without bloating overlay.py.

Could you please take a look if you get a chance, and check MCP3008 still works?