bverc/retropie-status-overlay #17: Compatibility with the PiJuice Hat

cainkirisgai, 2020-12-11:
Hi I have a pijuice hat that i am attaching to my build and was wondering if there is anyway to pull the battery info from it to display onscreen with this script or if im just going to have to put the ads1115 back in and try to see what i can do with it. in the Pijuice Hat software it shows an i2c address of 14, and i can get the voltage and battery info from the battery I currently have connected to it.

bverc, 2021-02-06:
Sorry, not sure how I missed this issue. 

Not in the current implementation, but certainly should be possible. I'm not too familiar with the PiJuice hat, but if you can get the necessary data over I2C, then it is entirely possible. 

The long term plan is to support multiple battery measurement options, but without hardware to test this becomes difficult

Current plan is to merge the changes in from #16 then I can create a new branch for you to test. Alternatively you can always fork and give it a go yourself, and I will happily merge in providing it doesn't break other functionality.

cainkirisgai, 2021-02-07:
From what I'm reading it does pull from the I2c, but I'm not sure about how to go through with fork tbh, haven't done too much in terms of compiling or creating scripts just some minor button commands and switches

bverc, 2021-02-07:
Well good news is as this tool is written in python, there is no compiling required (except for what is needed for pngview). If you want to start learning python for the raspberry pi, this would actually be a very good project to get started.

I'm aware not everyone enjoys coding though, so in that instance you can assist by providing as much information you can so someone can pick this issue up, especially if they don't have the hardware to test on. 

Some information you can provide:
- Is there a datasheet or wiki for the PiJuice?
- Are there any required libraries that need to be installed?
- Are there any code examples someone could use to easily add?
- What features does the PiJuice provide and how do you expect this tool will use them?

bverc, 2022-05-05:
Update on this issue: I have added crude PiJuice support. Currently it uses the reported battery voltage by PiJuice, rather than utilising the build in battery percentage/charging status it can report, due it not fitting with the current way the adc modules are setup. It wouldn't be difficult to customise the code to do this however, if working with other ADCs was not required.

To get this to run, package pijuice-base must be installed via apt-get (or pijuice.py copied from https://github.com/PiSupply/PiJuice/tree/master/Software/Source). I2C also needs to be enabled from raspi-config.