bverc/retropie-status-overlay #22: Would compatibility with PiSugar be possible?

LTCharlton, 2022-03-03:
There is an I2C table in its documentation (towards the end of its GitHub) but I'm not sure how that works. If it's possible and not too hard what are some of the steps to implement it?

https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series

bverc, 2022-03-05:
This certainly would be possible, although without hardware makes it very difficult to test and verify. 

This is a similar issue to #17 . My plan was to add some sort of modular plugin system that allows connection of different battery sensing options, but this has been on the backburner.

You are welcome to fork and add the functionality you need, then I'll have a starting point to make into some sort of modular system.

LTCharlton, 2022-03-06:
Thank you for your input. I'll fork it and take a look. Thank you.

bverc, 2022-05-05:
Well I have added the modular adc support, and I have made a rough attempt at getting something together for the PiSugar. However, I cannot test it without hardware. Could you have a look at the pisugar-test branch, and see if you can get it going?

Check you have I2C enabled in raspi-config. 

In the config.ini, set `ADCType = PISUGAR`, and `ADCGain = 1`