bverc/retropie-status-overlay #31: More detailed info needed

MrVinyla, 2022-06-20:
Can you explain these parameters, nothing is written about this in "read me"
[image]

bverc, 2022-06-20:
Excellent point. I will find somewhere to put this information. 

ADCGain you can use to upscale the read voltage if you are reducing (I.E using a voltage divider) before entering your ADC. For example, if you are reading a battery that could be as high as 4.5V during charging, but reading using an ADC with an upper limit of 3.3V, then you might use a (100K/100K voltage divider to drop the voltage to within the correct range. You then need to set ADCGain to 2.0 to bring this up to the correct voltage reading.

The four values here dictate how charged your battery is, and will require some testing using your battery and setup.

VmaxDischarging = Highest voltage seen after charging, power removed. (100% icon)
VminDischarging= Lowest voltage seen when discharging, and will trigger shutdown if desired (usually left at 3.2V for LiPo batteries)
VmaxCharging  = Voltage once charging is complete
VminCharging = Lowest voltage seen when charging a dead battery. (Hopefully this is higher than VminDischarging)

Which ADC are you using?

The logger in the tool will help you get this values. I'd be interested to know what values you are getting.

I will keep this issue open to make sure we fix this.

MrVinyla, 2022-06-21:
Python 3.5.3
I see, thats the problem?

bverc, 2022-06-22:
Can't help you update Python I'm afraid, but I would not expect you need to compile it from source. 

Might be simpler to just get a newer image of RetroPie on there.

MrVinyla, 2022-06-27:
New Retropie with Python 3.7.3. Quick solution for this?

[image]

MrVinyla, 2022-06-27:
[image]

bverc, 2022-06-27:
If you are not using the install script, follow the manual install instructions in the README.

`sudo apt-get install python3-psutil`

MrVinyla, 2022-06-28:
well, I finally overcame it
steps i did:
A new Retropie was installed, then I tried to automatically install retropie-status-overlay, I got an error due to a lack of some modules. After installing the required modules, the icons were displayed incorrectly. Then I tried to install manually - an error occurred that could not find the adafruit-ads1x15 library. Okay, I installed it: "pip install adafruit-ads1x15". Cron did not start the program automatically if I used the following command: "sudo crontab -e", for me this command "crontab -e" worked, then I added a line at the bottom as in the manual. And it works !!!! Thanks for such a good and useful app. Now I am checking the battery level.

bverc, 2022-06-28:
Glad you got it working,

Do you recall which packages were missing during the auto install? I wonder if the apt-get update failed somehow. 

Interesting comment regarding crontab, I think either should have worked since it is owned by pi user and accessible by both pi and root. I will confirm that.