bverc/retropie-status-overlay #32: ADC Arduino

R41zan, 2022-10-11:
Trying to get my arduino board that deals with button input to read voltage on it's analogue pin.

Arduino Script:
[image]

Python Script: (file name: m32u4.py)
[image]

Config.ini file:
[image]

Overlay.log Output: 
[image]

is there anything else I have to update? im not getting the overlay to display anything, not even the rest of the items (bluetooth, wifi, sound)

Thanks

R41zan, 2022-10-11:
As soon as I put the battery option in config.ini to "False" and restart the service it starts to output this on the overlay.log:
[image]

And the overlay starts to work again

R41zan, 2022-10-11:
Modified python script to call the read function to be able to test output:
[image]

The output: 
[image]

bverc, 2022-10-12:
Away from my computer, but I think I've spotted the issue. Using `if __name__ == "__main__"` means that code will not run when run as a module (as the script tries to do) but will work correctly when run directly.

Remove that line and unindent all below by one tab.

bverc, 2022-10-12:
Also. Stop the service then run the overlay.py manually. You should get an error output from python.

R41zan, 2022-10-12:
Updated config.ini:
[image]

Output: 
[image]

bverc, 2022-10-12:
Damn another bug in the install script. Can you please change `Type` to `ADCType` in your config file?

R41zan, 2022-10-12:
> Damn another bug in the install script. Can you please change `Type` to `ADCType` in your config file?

It seems like it's now trying to get the state of the battery but i'm not sure i can get that from the arduino as the only thing i can read is the voltage

R41zan, 2022-10-12:
Never mind, the config.ini file was missing `# ADC Channel for ADCs which have multiple channels
ADCChannel = 0`

R41zan, 2022-10-12:
Got it to not faill completely and the led on the arduino is showing activity but i'm getting this error:
[image]

R41zan, 2022-10-12:
So Amixer error is unrelated at this point. 

What could be causing the overlay to not display the battery icon even after getting the data? Data format? Because at this point it should be getting a value to the script from the function read()

bverc, 2022-10-12:
Great. Getting closer.

Looks like amixer cannot find your audio device. Can you try running `amixer` in the command prompt and dumping the results here?

Also try `amixer -c 1` and `amixer -c 2`

In the mean time. I suggest setting Audio to false while we try to get battery working.

bverc, 2022-10-12:
I think the program is just getting stuck once it gets to audio. Disabling that should make the others work.

`python3 overlay.py` should be putting out data every few seconds when working correctly.

R41zan, 2022-10-12:
Put a print on the python script to see what data if being returned by the read() function

[image]

R41zan, 2022-10-12:
Audio is set to false and no longer gives a error
Still no battery icon or info

[image]

doesn't seem to go from 1 line of display info

R41zan, 2022-10-12:
Am i sending data too fast repeatedly?

R41zan, 2022-10-12:
to be clear the bluetooth and wifi icon still work fine

bverc, 2022-10-12:
When you say bluetooth and wifi work fine, do they change if you move out of range for example, or just the same as when they first load.

There are a few more errors in your config file which may be causing it to get stuck, I'm not sure.

Try `VMaxCharging = 4.7` since you are returning higher vslues than I would expect.
Also `ADCShutdown = False` not `N`

R41zan, 2022-10-12:
I've updated my config.ini to the config.ini.example and re-did it so it should be fine now. 

Bluetooth changes state when i connect somethig and wifi changes signal strengh

ill try with higher VmaxCharging

bverc, 2022-10-12:
Please also share your m32u4.py

R41zan, 2022-10-12:
No changes. It doesn't change icon behavior

also updated config.ini:
[image]

R41zan, 2022-10-12:
File m32u4.py
`
#!/usr/bin/env python3

#Import necessary library
import serial
import time

def read(channel):
    ser = serial.Serial('/dev/ttyACM0', 9600, timeout=1)
    ser.reset_input_buffer()
    while True:
          value_v = ser.readline().decode('utf-8').rstrip()
    return value_v
`

R41zan, 2022-10-12:
[image]

bverc, 2022-10-12:
Aha! Ok it's getting stuck in your while loop.

Remove `while True:` and pull in the next line.

`while` True` will never exit.

R41zan, 2022-10-12:
Alright, something different:

[image]

R41zan, 2022-10-12:
i probably shouldn't be opening a serial connection everytime the function is called, right?
move the serial out and only have the buffer reset and value_v in the read()?

R41zan, 2022-10-12:
I have battery info!!!!

R41zan, 2022-10-12:
that was the issue. 
Fixed it by changing 
`return value_v` to `return float(value_v)`

bverc, 2022-10-12:
RuntimeError will be fine. Just copy the other module.

bverc, 2022-10-12:
Which delay are you referring? Let the main script determine the polling frequently not your module.

R41zan, 2022-10-12:
On the arduino side i have a 500 delay

bverc, 2022-10-12:
Actually that is a good point. Overlay will pull data from the buffer slower than the arduino puts it on. Which means without clearing the buffer before a read it could be pulling old data.

I suggest reducing delay so something like 100, but then clearing the buffer before doing a read to request the latest value. 100ms means it won't need to wait too long.

To test, put your battery on and off charging and check you get immediate response in the overlay log voltage reading (not the icon as this uses a rolling average)

Hope that makes sense.

R41zan, 2022-10-12:
[image]

Tested no delay on serial send from the arduino. 
The quantity of info sent must overflow/"mix" with the quantity in the buffer and eventually it craps itself. Also note that theres a couple of voltages that are anormal - 18 and 3 in this case

R41zan, 2022-10-12:
def read(channel):
    ser.reset_input_buffer()
    value_v_string = ser.readline().decode('utf-8').rstrip()
    value_v = float(value_v_string)
    return value_v

my read() is currently like this so when it's called it should clear the buffer and read a new line sent by the arduino
at some point this must cut in the middle of some bytes being sent by the arduino and convert the bytes to utf-8 in a way that creates abnormal numbers (?)

R41zan, 2022-10-12:
Arduino side: 
[image]

Will change serial print to write to send bytes instead of readeable data to see if it helps with consistency

R41zan, 2022-10-12:
nvm that would complicate things quite alot

bverc, 2022-10-16:
Added a modified version of the python module to the branch `serial-test`. Can you please test it works on your setup?

scavenrage, 2024-03-19:
Hi, can you please post the final m32u4.py? I'm trying to follow your comment to have the final version but I'm still not getting the battery icon... 
Thanks

scavenrage, 2024-03-19:
Oh nevermind, I've just succeed to have the battery info! the final version of the file is

#!/usr/bin/env python3

#Import necessary library
import serial
import time
ser = serial.Serial('/dev/ttyS0', 9600, timeout=1)
ser.reset_input_buffer()
def read(channel):
    value_v = ser.readline().decode('utf-8').rstrip()
    return float(value_v)

bverc, 2024-03-19:
Hi @scavenrage 
Did you test the `ser` module on the `serial-test` branch?

You code looks very close to what we arrived at in `ser.py`, except you are using `/dev/ttyS0` instead of `/dev/ttyACM0`. Are you connected via USB or some other connection?
I assume this would therefore need to be customizable? Although seems like /dev/ttyACM0 is more correct if using USB:
https://forums.raspberrypi.com/viewtopic.php?t=207893
https://arduino.stackexchange.com/questions/19648/difference-between-dev-ttyacm0-and-dev-ttys0-arduino-ide-ports-under-linux

scavenrage, 2024-03-20:
Hi,
I used ttyS0 because I'm not using USB but I'm connected directly with the gpio15 that is the rxd of the pi (in this case it's a CM4 but it's the same).

I'm sorry but I didn't test the ser module, only because I didn't know how to do that! I just found out now how a branch works. Tomorrow (here it's 2am) I will come back with the result

bverc, 2024-03-20:
Makes sense.
Yea give it a go, and if you are happy it works, I can merge it in. It was never confirmed as working previously, and I was unable to test.

As the module can't read from config file, the SERIAL_DEVICE can't be set outside the module. Open to ideas how to do this. Can add some comments in the module to offer examples.

scavenrage, 2024-03-20:
Hi bverc,

just finished now testing the ser module, works like a charm!  I  had just did a correction, because without that it does not work
the correction is (as saw in the discussion before) to change return value_v to return float(value_v).
And of course I've changed the serial port because i used the ttyS0 (it works also putting "serial0" instead of ttyS0).

Sadly I don't have any good idea to set the serial device other than just writing it in the module after searching for the serial ports available with the command "dmesg | grep tty"

[image]

bverc, 2024-03-21:
Great, glad it is working for you. I'll get that line tidied up and merged in. Cheers for testing it out.

commit ca237220 (2024-04-09): Changed read function to return float of value

Fixes #32
commit 40f73ee7 (2024-04-09): Add serial device for ADC (#36)

Fixes #32
closed 2024-04-09 by bverc