bverc/retropie-status-overlay #34: FPS drops

Guma53, 2023-06-19:
When using ADS1015 or INA219 ADC, i got FPS drops on all games e.g. on GB emulator, from 60 to 35-45 FPS every 5 seconds. Is it related to logging? Is there any think i can do with it. Running on RPI zero W, with gpio 2,8' 320x240 LCD.

bverc, 2023-06-20:
The main loop runs every 5 seconds, so this would seem likely. Not sure it would be specific to logging. Probably just one or more of the modules are really inefficient.
Which icons do you have enabled? If you disable battery icon, does it return to normal?

Guma53, 2023-06-20:
It doesn't matter what icon is enabled, always lagging. It return to normal only after 'service retropie-status-overlay stop'