bverc/retropie-status-overlay #38: Must the ADC be an ADS1015?

voxpop9, 2024-06-18:
I currently have an ADS1115 on hand, i am wondering if it is compatible, or do I have to buy a new ADS1015?

bverc, 2024-06-19:
Shouldn't need to. Plugin will just need to be tweaked slightly.
This code was inherited from [d-rez/gbz_overlay](/d-rez/gbz_overlay), and I don't think anything has been tested on actual hardware recently.

Create a copy of [adc/ads1015.py](/bverc/retropie-status-overlay/blob/master/adc/ads1015.py)  to `ads1115.py` then change:
`adc = Adafruit_ADS1x15.ADS1015()`
to
`adc = Adafruit_ADS1x15.ADS1115()`

Then the following two rows will need to be edited to something that makes sense. If using gain 2/3, then the bit size reduces to 187.5uV:
```
value = adc.read_adc(channel, gain=2/3) # Channel from config.ini, 2/3 gain
value_v = value * 0.0001875 #1 bit = 187.5uV
```

Once you have that setup, you should be able to choose ADS1115 in you config.ini file:
`ADCType = ADS1115`

Like I say, I'm not sure this plugin was ever tested, so keen to hear how it goes.

voxpop9, 2024-06-27:
I am running the ADS1115 into a logic-level converter to use 5V to read the battery voltage directly. The issue is that turning off the Pi would cause it to back drain through the ADS, so I added a 300kOhm resistor in series to bring that current down to 10 microamps. The ideal is to use a switch to cut the connection but I did not have the time to do that (currently overseas with my Portable RetroPie Project).

voxpop9, 2024-06-27:
sudo service retropie-status-overlay start is not working, I installed it using automatic installation, running custom config file. System reboot terminal shows `overlay-daemon` failing to start.

voxpop9, 2024-06-29:
Nevermind, i got it to work. Turns out I forgot to enable I2C :P

bverc, 2024-06-30:
Glad you got it working. Did you get the ADC driver to work?
If you can post your working code, I can add to to the repository. Alternatively, create a pull request.

voxpop9, 2024-07-06:
> Glad you got it working. Did you get the ADC driver to work? If you can post your working code, I can add to to the repository. Alternatively, create a pull request.

Apologies for the late reply, indeed it is working. Your solution does work though i had to add a resistor between battery and adc rad pin 0 to prevent back voltage when off. A modification im going to do is have a switch to disconnect the battery from the ADC physically to remove the voltage drop across the resistor.

closed 2024-07-07 by bverc

bverc, 2024-07-07:
Excellent, thank you for confirming it works. I have committed the changes to master, so please check if you are able.

Awesome build! Looks very slick. 

I think the best solution to connect battery to the ADC is via a PFET. Refer this link with an example circuit:
https://electronics.stackexchange.com/questions/337058/low-current-divider-for-battery-voltage-monitoring

voxpop9, 2024-07-09:
> Excellent, thank you for confirming it works. I have committed the changes to master, so please check if you are able.
>
> Awesome build! Looks very slick.
>
> I think the best solution to connect battery [...]

So basically, have the battery hooked to a P-channel MOSFET that is controlled by the Pi that enables the MOSFET everytime a measurement is made?

bverc, 2024-07-09:
No, I think that is more complicated than it needs to be. Assuming your 5V rail is after some sort of power switch, just replace GP_OUT with 5V. When your device is on, 5V will turn the BJT on, turning the PFET on and providing a voltage to your ADC. When device is off, BJT will be off, PFET will be off and ADC is protected.

voxpop9, 2024-07-09:
Currently, my battery to ADC does not run through a Voltage Divider as I am powering the ADC with 5V from the Pi 5V GPIO pin. There is a logic level converter to convert the ADC signals from 5V down to 3.3V. Im looking at the circuit and if I read correctly, a PFET is preferred when going from +ve to Load compared to an NFET since if I use an NFET, I would need to provide a higher voltage to turn it on. However, can I not just then hook up the 5V rail to an NFET that is connected between battery and ADC since 5V is higher than the batteries 3.7 nominal, 4.2 max?

bverc, 2024-07-09:
I think I see what you are proposing. However, I don't think it would work without an "ideal" NFET.

To get current to flow Drain to source, you would need a pull up resistor to 5V. While 5V rail is live, NFET drain would be held Vds above battery voltage, and when 5V is off, NFET turns off. ADC input is now floating as 5V is no longer live.

Would the voltage on the battery not be higher than 4.2V while charging? The default config assumes a Vmax_Charging of 4.5V, and I imagine it could be higher.

You could have a NFET with a Vgs < 0.5V, but at this voltage, the NFET will not be in saturation and Vds will be very high (in the volts).  See this example from one of the few NFETs on digikey with a Vgs of less than 0.5V, Vds is over 4.5V. You would need a Vgs of 1.5V before the Vds is negligable.
[image]

You might be able to find a close to ideal NFET to suit your needs, but I wouldn't be hopeful. The solution I linked, although slightly more complex, is probably quite a bit more reliable.

voxpop9, 2024-07-09:
I see, Thanks for teaching me this, reading up has taught me some stuff about FETs and BJTs. I do have 1 solution in mind which is a lot simpler although probably dumber: I could just have a separate physical switch between Battery and ADC. This switch will be placed physically next to the power switch, with a 3D printed switch to physically join the two Hahaha. 

Just to make sure, From the circuit, it looks like it is using a BJT that turns on when it is powered by 5V, which then drains the PFET,   allowing battery to then connect with ADC? Wouldn't there be some power loss from the BJT and PFET?

bverc, 2024-07-10:
Yes your idea solving this mechanically is a great idea. Doesn't even need to be two switches. You can get a Double Pole (DPST or DPDT) switch, which would connect the two circuits simultaneously without the need of a second physical switch. 
One pole to enable the supply, while the other connects the battery to the ADC. Isolated from each other.

You are right, there will be some current loss from the circuit while the BJT and PFET are both on, but this would be negligible compared to the power pulled by the device. And neither should be powered while the device is off, so no risk of current leaks while off.  2 * 5V^2/100K ~0.5mW. The Raspberry Pi Zero pulls 400mW, so 3 orders of magnitude lower. 

If you prefer the mechanical solution over the electrical solution, it is perfectly acceptable.

voxpop9, 2024-07-10:
Also, why not just use the BJT directly as the switch? Why use it to turn a Mosfet on?

bverc, 2024-07-10:
The BJT (could also be an NFET) acts as an inverter, switching on the low side. It converts logic high (5V rail) to logic low (0V) to turn on the PFET on the high side. 

If you just switch on the highside only, you would need a logic low to tun on the PFET. You could do with a GPIO on the Raspberry Pi, but then you would want it to be logic high (VBAT) when the power is off.  

You could do this with a GPIO that is set to low in config.txt and goes high impedance when off, which could then be pulled up to VBAT with a resistor. The issue with this approach is you WILL get some current leakage while the device is off, and also you are relying on the Raspberry Pi GPIO circuit to have a high impedance, when it likely has a weak pull up/down.

Possible, but can't be certain of all consequences. Battery loss over time would be one.

voxpop9, 2024-07-10:
I think i may have phrased my question wrong, What i meant is why not use the BJT only between the battery and the ADC?

bverc, 2024-07-10:
If you are still stuck with your circuit. Maybe create a new issue, and keep this one dedicated to ADS1115 support.

voxpop9, 2024-07-10:
> If you are still stuck with your circuit. Maybe create a new issue, and keep this one dedicated to ADS1115 support.

Aah I see, My apologies and thank you for all the help you gave.

bverc, 2024-07-10:
Not a concern. Happy to assist.

commit 9d336e5b (2024-07-07): Added ADS1115 Support

Fixes #38